min-protocol/min #2: Code maintainence

Resonanz, 2017-03-06:
HI Ken, are you still maintaining this code? Thanks.

kentindell, 2017-03-09:
I've got a big release to come based on experience of using MIN in a project. The changes include:

- Support for bigger frames (up to 255 bytes)
- Move to 32-bit CRC (based on experience with VCP USB - surprisingly noisy)
- An optional sliding window transport protocol layer (i.e. reliable connection between host and target) with Python implementation for the host side.

The protocol still scales from low-end 8-bit devices like the AVR (I developed it first for an Arduino board) but I've been using it on a Cortex M0. I've now got to find the time to package it up for a release.

Resonanz, 2017-03-09:
Great, thanks Ken.  Any idea of a time frame (bad joke, I know).  But seriously, I was about to write code for doing this and we can certainly give you some feedback on the new version when you release it.

kentindell, 2017-03-11:
I'll work on it today :-)

kentindell, 2017-03-11:
Major new release. Have a look at the code: please raise issues with questions.

NB: this has not been re-tested in a live system yet. It was cut out of my custom Pyboard firmware (which uses USB VCP) and I haven't had a chance yet to roll this back in and check it still works after the refactoring.

closed 2017-03-15 by kentindell