min-protocol/min #3: Python 3 on Linux

Resonanz, 2017-03-06:
I think I have the Python part of this working on Ubuntu Linux under Python 3. Should I send instructions, or do a pull request, or just post you the details?

kentindell, 2017-03-09:
Err.. good question. I should probably release the new MIN first so you can see what changes there are for the transport layer.. (it's currently running in Python 3 on both Windows and Ubuntu. It's thread-safe and uses Pyserial).

Resonanz, 2017-03-09:
Oh, okay, sounds good. It took me a few minutes to get the code ported to Py 3 on Ubuntu. So yes, please release the new version and we will test it for you :-)

kentindell, 2017-03-11:
Looking forward to test results :-)

Any questions and I'll update the code to include comments.

Resonanz, 2017-03-12:
Hi Ken, Thanks a lot for the update !  I think the README needs updating also -- it is still at 1.0.  

Couple of questions to address in the README:

1. What are you compiling on? Atmel Studio? Arduino? 

2. What code is needed to make this work?  Is it just one header file and one c file?

Thanks !

Resonanz, 2017-03-12:
Hi again Ken, would it possible to also add a picture or text of the packet shape into the README, e.g.

ID :: NUMBYTES :: ...... :: CKSUM

ID = ...
NUMBYTES = total bytes in packet including ID and CKSUM bytes
...
CKSUM = 32 bit CRC...

kentindell, 2017-03-12:
Here's the frame diagram, in the wiki:

https://github.com/min-protocol/min/wiki/Frame

kentindell, 2017-03-12:
And I've edited the version number in the README now :-)

The code now doesn't require tight binding to a driver: it uses callbacks, and your own code calls your own serial drivers. This makes it a lot easier to work with. You just have to compile the C file and link it in to your project (you might want to put it in a library but the code is so simple I would just add it into your Makefile or whatever you use).

Your application just needs to include min.h to access the API.

I cut this code out of my custom Micropython firmware for the Pyboard, which runs on a Cortex M4 device. But it ought to work on an Arduino or anything else: there's nothing special to the C (I guess the new style // comments might barf on a very old compiler). The code is written for readability but with an eye to how a decent compiler should spot what's happening and create good code.

I haven't had a chance to compile this up as a standalone project yet: I need to carve out a test build where I can run Python and C back-to-back. I was thinking of doing this against the Pyboard, but I guess an Arduino could be used.

Resonanz, 2017-03-12:
Thanks a lot Ken. The code looks really nice and well thought out...

kentindell, 2017-03-12:
I hope it works well for you. Feedback/issues/questions very welcome.

closed 2017-03-15 by kentindell