min-protocol/min #4: Layers

Resonanz, 2017-03-13:
I think it would be helpful if the layers that are referenced in the WIKI and elsewhere were listed and defined.  For example, I see there is a **Layer 0** -- but it is not clear whether this is this part of the 7 layer OSI model? The **Bit** layer is presumably the **Physical** layer in the OSI model.

Thanks!

kentindell, 2017-03-13:
I should probably stop talking about the layer stuff: I did think about making frames and transport connections separate layers but I rolled them together to save bandwidth (the Sequence field isn't included if the top bit of the ID/Control field is 0, which isn't the way to do proper layering). A bit hacky and not as clean as TCP over IP but hey, at least I didn't use 32-bits for addresses that are going to run out..

Resonanz, 2017-03-13:
LOL, that made me laugh.  At least IPv4 and 192.168.1.1 is easy to remember.

So layers, well, I guess a diagram in the Wiki would be helpful...

closed 2017-03-15 by kentindell