min-protocol/min #5: Arduino compilation

Resonanz, 2017-03-13:
Hi Ken, I have tried to compile the min.c file inside an Arduino environment. To do this I have simply `#included` the min.c file (which itself `#includes` min.h). The compilation is perfect except for the following:

`In file included from C:\Users\resonanz\Documents\Arduino\MIN_testing\MinTest\MinTest.ino:25:0:

C:\Users\resonanz\Documents\GitHub\min\target\min.c: In function 'void rx_byte(min_context_t*, uint8_t)':

C:\Users\resonanz\Documents\GitHub\min\target\min.c:443:47: warning: left shift count >= width of type

             self->rx_frame_checksum = byte << 24;

                                               ^

C:\Users\resonanz\Documents\GitHub\min\target\min.c:447:48: warning: left shift count >= width of type

             self->rx_frame_checksum |= byte << 16;

                                                ^ `

`byte` is an uint8_t and so I guess the compiler is having a hissy about the shifting more that 8 bits.  To test, I replaced the 24 and 16 above with 8 and everything compiles fine.

kentindell, 2017-03-13:
That's me being sloppy: there's an implicit assumption integers are 32 bits. Bad boy!

Needs to be cast to a 32-bit integer first before shifting.

You can't change the shifts: it will all explode. Basically this is assembling a 32-bit number byte-by-byte. The compiler is normally smart enough to work out what's going on. Let me fix it.

kentindell, 2017-03-13:
https://github.com/min-protocol/min/commit/0a202975b3a1e9400541a64f1c1bb3868a02c487

closed 2017-03-13 by kentindell

Resonanz, 2017-03-13:
I would rewrite this as follows using almost identical code for each additional byte (is there a better way?)

```
        case RECEIVING_CHECKSUM_3:
            self->rx_frame_checksum = byte;
            self->rx_frame_checksum << 8;
            self->rx_frame_state = RECEIVING_CHECKSUM_2;
            break;
        case RECEIVING_CHECKSUM_2:
            self->rx_frame_checksum |= byte;
            self->rx_frame_checksum << 8;
            self->rx_frame_state = RECEIVING_CHECKSUM_1;
            break;
        case RECEIVING_CHECKSUM_1:
            self->rx_frame_checksum |= byte;
            self->rx_frame_checksum << 8;
             self->rx_frame_state = RECEIVING_CHECKSUM_0;
            break;
        case RECEIVING_CHECKSUM_0:
            self->rx_frame_checksum |= byte;

```

kentindell, 2017-03-13:
You could do that but on an 8-bit device it means shifting a 32-bit word, which is fairly costly (it might even try and do something like four individual rotates in a loop executed 8 times). Good compiler optimizers are generally coded to see things like "<< 16" and "<< 24" and see that it's really just writing a byte to the nth byte of a word and then optimize the code to do that directly.

Need to have a look at the assembly language to see what the Arduino compiler makes of it. GCC on the ARM is very slick at this kind of thing.