min-protocol/min #7: A bit stuck

Resonanz, 2017-03-14:
Hi Ken, it has been a frustrating day.  Arduino is C++ so it took quite a bit of effort to get the C code to compile.  I got there eventually using `extern "C"` to wrap your header file. There is now some serial activity but it is not working.  I am really having problems with the steps I need to go thru to get things up and running.

The Python code also seems to be different to the previous version.  I installed Continuum's Anaconda with Python 3.6 (I realise you specify 2.7, but figure it would likely work with a little tweaking). Not so far however -- the code just runs and exits:

```
C:\Users\resonanz\Documents\GitHub\min\host>python min.py

C:\Users\resonanz\Documents\GitHub\min\host>
```

I'm sure there is more to come :-)

Resonanz, 2017-03-14:
These are my definitions so far:

```
min_context_t *mct;
uint8_t payLoad[5] = {1, 2, 3, 4, 5};
min_init_context(mct, 1);   // (context, port)
min_send_frame(mct, 0, payLoad, 5);

```
And these two functions are callbacks I think.
```
void min_tx_byte(uint8_t port, uint8_t byte) {
  Serial.write(byte);
}

void min_application_handler(uint8_t min_id, uint8_t *min_payload, uint8_t len_payload, uint8_t port) {
  Serial.print("got something");
}
```
It would be really helpful to have some further documentation so we can get this going. I am hoping to have some LabVIEW code written to communicate with the C code -- I think that is close to completion.

Many thanks.

kentindell, 2017-03-14:
Firstly, I'd really not recommend #including C code into a file: it's a really bad style. You should add min.c into your project (do you use a Makefile or does your IDE build one for you?). Then the compiler knows it's C and not C++ and compiles accordingly. Also this way you can set your own compilation options for just that file.

kentindell, 2017-03-14:
Secondly, the Python code is a module, not a program: it's intended that you import the module into your own program and call it from there. So it definitely won't run by itself. You could easily make a trivial main.py program to call it for testing and that's where I'd start: a "print everything you see" program.

kentindell, 2017-03-14:
Thirdly, you haven't actually created a MIN context variable there: you just created a pointer to one that's uninitialized. You should do something like:

    min_context_t mct;
    min_init_context(&mct, 1);   // (context, port)
    min_send_frame(&mct, 0, payLoad, 5);

The code you had would've been dereferencing a pointer to somewhere random and this would go bang. The reason the API expects you to do the memory allocation for the MIN context is that the code doesn't (and shouldn't) use any heap or dynamic memory: only the programmer knows the lifetime of the memory. Oh, and don't make the MIN context a local variable in a function: when you return from the function it will be deallocated from the stack and when you later call MIN you'll get your stack corrupted by writing to where that context was on the stack.

The two callbacks you have look fine.

kentindell, 2017-03-14:
Oh, and I forgot to say: the Python I wrote is Python 3 (I don't think anyone should be writing for Python 2 any more).

Resonanz, 2017-03-14:
Hi Ken, it seems that Github doesn't support threaded comments, so I'll follow the order above.

**C inside C++**
The way this is done on Arduino is to use the extern keyword. So in fact, I don't have to touch your .c and .h files at all and simply need to use `extern "C" {#include min.h}` in my Arduino code. It took a while to get this working and lots of people has asked how to do it online. Accessing a makefile may be possible but it is well hidden.  Perhaps we should move to Atmel Studio, but then we lose the advantages of quick development (our stuff is not commercial).

**Python**
Okay, got that. Re the version, I think the helper file says python 2.7 (perhaps that is the min version  -- bad pun sorry).

**MIN context variable**
LOL, the code you show is what I was using, until I changed it. Thanks for clarifying.  The `struct` is a global in my code.

Resonanz, 2017-03-15:
Hi Ken, to help us get started, is there any chance that you have the basic Python code available that uses your library?  Or is this documented somewhere? Thanks.

kentindell, 2017-03-15:
I've got a very complex host monitor program that talks to the Pyboard to use the Pyboard as a CAN interface. I can try and cut something out of it and show you how it works but I haven't got time today to test it actually works.

kentindell, 2017-03-15:
https://github.com/min-protocol/min/commit/41afd8d347cb98b359f800d3cb89f6876a9a0dfa

closed 2017-03-15 by kentindell

kentindell, 2017-03-16:
Does that work for you?

kentindell, 2017-03-20:
I've added a "hello world" program (it really does say "hello world") that consist of a simple Python program and a sketch for the Arduino. More here: https://github.com/min-protocol/min/wiki/Running-the-example-program