min-protocol/min #8: min_time_ms()

Resonanz, 2017-03-14:
Hi Ken, I think this callback function documentation needs some more info. This is called by your code but I'm not sure what it is passing...?

```
min_time_ms()
This is called to obtain current time in milliseconds. This is used by the MIN transport protocol to drive
timeouts and retransmits.

```

kentindell, 2017-03-14:
This returns the passage of time in milliseconds. It's used to drive the timeouts in the transport protocol. You need to return a 32-bit integer for the current time in milliseconds. You probably have a periodic interrupt that increments a variable every interrupt? If not then you need to synthesise a timer somehow: maybe read a free-running counter and convert to milliseconds? It depends on your run-time environment.

Resonanz, 2017-03-14:
Okay, thanks!  That's easy to implement. So MIN calls the function and gets a signed or unsigned int?

kentindell, 2017-03-14:
uint32_t: an unsigned int.

closed 2017-03-15 by kentindell