min-protocol/min #12: Was python2 phased out?

mcondarelli, 2018-05-04:
Apparently MIN needs python >= 3.5 to work (due to "type hints", at least).
Is this intentional?
I know we are in end-of-life for python2, but it is still widely used (and installed by default in many distributions), Installing python3 just to have MIN working seems a bit excessive, unless there are other motivations.

kentindell, 2018-05-04:
Python 2 is the IE 6 of programming languages. Time we all moved on ;-)

mcondarelli, 2018-05-04:
Unfortunately all distribution maintainers (from Debian to RH and beyond) hold a different opinion.
Not to speak about some major packages (Calibre comes to mind).
MIN is meant to be used in small space (at least the "server" part) forcing to install a full python3 (while you already have python2!) to use it seems a bit of an overkill.
Anyways, if the answer is "this is intentional, we want to help phase-out python2" I'll accept it.
I just wanted to be sure about motivations.
Thanks.

kentindell, 2018-05-04:
The reason is exactly as you say: I do want to help phase out Python 2. 

You make a good point about type hinting: I use Pycharm and find type hinting support really excellent (it does an excellent job at tracking types and flagging bugs). But I found out recently macOS supports Python 3 as standard but not new enough for type hinting and it is silly to force an install of a new Python interpreter just for that. I'll take out the type hinting.

As for the resources required for MIN: I built a Python implementation because it's very common to run a PC at the server end (I use it with a GUI to control data capture from a Cortex M4 device). But there obviously needs to be more implementations to fit with PC environments better, as well as for higher end embedded devices (e.g. Raspberry Pi).

mcondarelli, 2018-05-04:
Thanks.
I'll close this.
FYI: I'm going through the loops of backporting to Python2.7.

closed 2018-05-04 by mcondarelli