min-protocol/min #16: Question about sequence numbers

mikemoretti3, 2018-11-06:
Hi,

I'm trying to use the MIN protocol for a board that talks to 4 other boards via RS485 using one UART (with 8 RS485 transceivers, 4 on the control board and one on each sub board).  Unfortunately, the control board sequence numbers increment 4 times faster than the sub-boards and so any message after the first ACK from the control board will not be handled by any board except the first.  Is there some reason you used '==' for the sequence number check instead of >=?  From what I remember, TCP does it that way (i.e. it doesn't care if the sequence number is equal but only if it's >=).  I'd prefer not to have to use 4 min_contexts for my comms.

Thanks!
-m

robthepyro, 2020-10-13:
@mikemoretti3 did you ever come up with a solution to this? I'm also playing around with MIN and RS485, trying to work out how to deal with the half-duplex flow control aspext

mikemoretti3, 2020-10-13:
no, i gave up on this protocol and developed my own.  but this problem only occurred when talking to multiple boards from one main board.  from your comment it doesn't sound like that's what you're doing?

kentindell, 2020-10-13:
I designed MIN for MCU-host communications rather than a multidrop bus because there wasn't anything simple to do the job. But if you want a multidrop bus protocol there are other options. Have you come across [LIN](https://en.m.wikipedia.org/wiki/Local_Interconnect_Network)?

robthepyro, 2020-10-15:
> 
> 
> no, i gave up on this protocol and developed my own. but this problem only occurred when talking to multiple boards from one main board. from your comment it doesn't sound like that's what you're doing?

Yeah I might end up doing my own protocol, I was actually just trying to not re-invent the wheel haha. I actually have a couple of devices on a single bus (that control servo motors) and a main controller. Really the system can be one direction half duplex controller -> slave, which will help with arbitration. I'll likely do something simple like $$"node id"*"len"#"payload ascii only"@"CRC"/n 
The system is critical, so I need all the error checking capability, very low bandwidth however so even MIN is a bit over the top. 

> 
> 
> I designed MIN for MCU-host communications rather than a multidrop bus because there wasn't anything simple to do the job. But if you want a multidrop bus protocol there are other options. Have you come across [LIN](https://en.m.wikipedia.org/wiki/Local_Interconnect_Network)?

You are correct, MIN is awesome  and I've actually used it for another project before in the past with great success, but not ideal for this application actually. LIN is actually a brilliant idea, never used it before but could fit the bill well. Also likely to be some good libraries out there to same me some time.