min-protocol/min #28: Question regarding the signal layer json

Fladdan, 2021-03-30:
Hi @kentindell 

I'm currently having a project, where I want to use your nice min protocol!
I have an **PC** which is communicating over **UART** with an **uC** and that **uC** is communicating with **another uC** over **RS485**. I'm using a windows PC and  STM32 m4 and m7 controllers.

The example from this repo is already working between STM32 and the PC. 

My question:
Would it be possible for you to upload an example of the implementation on the Arduino & python side with the signal layer in the json format? Like the example you posted in your blog: https://kentindell.wordpress.com/2015/02/18/micrcontroller-interconnect-network-min-version-1-0/.

It would be nice to get an working example of how to implement the generated .c and .h files from the python code generator you wrote.

Best regards and thank you for the work you have done on this open source project,
Daniel

kentindell, 2021-03-30:
I didn't get too far with the signals layer and code generator. I'm going to come back to this soon because the cantools library for Python does pretty much exactly this but for CAN, so I'd like to interface to that project (MIN was inspired by CAN and LIN and the concepts are the same).

https://cantools.readthedocs.io/en/latest/

Fladdan, 2021-03-30:
Thank you for your answer :) and greetings from Austria. 

I will have a look if the documentation and code from the cantools library **can :D** help me.