min-protocol/min #33: Question: Is Data-Streaming Supported?

oswinrodrigues, 2021-07-08:
This ticket is a question, not a bug or feature request.

I'm interested in whether I can use a data-streaming pattern (as opposed to a request-response one) with this library. Hoping to have a sensor just continuously send over measurements, without the MCU having to explicitly ask for it each time.

Searching around, I noticed the "open loop" [design objective](https://github.com/min-protocol/min/wiki#design-objectives) which might suggest that this _is_ supported. The same with this [example code](https://github.com/min-protocol/min/blob/master/target/sketch_example1/sketch_example1.ino#L113). Just looking for confirmation to be 100% sure.

Thanks!

kentindell, 2021-07-08:
Yep, for sure. You just push messages in and they come out the other end. The transport layer on top adds reliable transmission if there is noise.

oswinrodrigues, 2021-07-08:
Awesome, many thanks!

closed 2021-07-08 by oswinrodrigues