min-protocol/min #37: Any port for C#?

eriknieel, 2022-04-26:
Hi @kentindell 
Will you make a demo for C# in the future? I think that C# is closer to C than Python so it's easier for newbies to understand each part by learning the host/target demo.
I hope you would consider that soon.

Anyways, your work is awesome, thank you. You're my inspiration!

kentindell, 2022-04-26:
I haven't written C++ for years. These days I use C for embedded and for performance features (then wrapped in Python) and Python. I've actually wrapped the MIN C code in MicroPython on the Raspberry Pi Pico.

Python and C is where it's at! :)

eriknieel, 2022-04-27:
I see. Thanks for your magic. Best wishes and please do more awesomeness! :p

closed 2022-04-27 by eriknieel