min-protocol/min #40: Received spurious ACK

milo1000, 2022-12-03:
Scenario for this problem:
1. Host (using min.py) sends some frames to client (which is Arduino with min.c)
2. Arduino listens for incoming frames and sends "keep alive" frames to notify host that everything is OK
3. At that point Arduino client is restarted (hard-reset)
4. Host keep sending ACK frames for some old seq, client prints "Received spurious ACK" and basically stops working
5. Due to lack of "keep alive" frames, Host decides to perform transport_reset()  - that resets client, but unfortunately Host keeps sending "spurious ACK"
To fix that I've added **self._rn = 0** to def _transport_fifo_reset, it seems to work OK, but need to be confirmed by someone with expertise.
```    
def _transport_fifo_reset(self):
        self._transport_fifo = []
        self._last_received_anything_ms = self._now_ms()
        self._last_sent_ack_time_ms = self._now_ms()
        self._last_sent_frame_ms = 0
        self._last_received_frame_ms = 0
        self._sn_min = 0
        self._sn_max = 0
        self._rn = 0
```

kentindell, 2022-12-29:
Yes, that should include that (it mirrors the embedded side). Can you make a Pull Request? I can then accept it.

subodh-malgonde, 2023-01-06:
Looks like this issue has been fixed https://github.com/min-protocol/min/pull/42

schperplata, 2024-12-22:
I am still getting the exact same `Spurious ACK` issue.  @milo1000, did the fix #42 resolved your issue?

milo1000, 2024-12-23:
Yes, it did.