min-protocol/min #46: Multiple Ports

lobsterdude, 2023-05-23:
Using min.c for multiple contexts (multiple calls to min_init_context with different port numbers and multiple calls to min_poll), I get cross-port data corruption due to the shared payload buffer "payloads_ring_buffer". If I tie payloads_ring_buffer to the min context struct, these issues are solved.
Am I missing something or is this a bug ?

kentindell, 2023-05-23:
The implementation isn't designed to support multiple ports (it's designed to fit into tiny microcontrollers to talk to a host) but it could be: by moving all the global variables (including buffers) inside a struct for each port and then parameterising the calls with a pointer to the port struct (as suggested).

lobsterdude, 2023-05-24:
Great, thanks for the fast reply. The presence of the port variable was a bit misleading, now I know.

closed 2023-05-24 by lobsterdude