min-protocol/min #48: Memory conflict: Multiple MIN ports access the same payloads_ring_buffer

kinimodkoch, 2023-07-17:
There seems to be a bug if you use multiple MIN ports at the same time with the TRANSPORT_PROTOCOL option enabled.

There is only one ring buffer defined in min.c:

```
// Where the payload data of the frame FIFO is stored
uint8_t payloads_ring_buffer[TRANSPORT_FIFO_MAX_FRAME_DATA];
```

While all other port related data - including access indices for this buffer - are available for each port in

`struct min_context`

The effect is that each port overwrites buffer data of the other ports.

I suggest that
`uint8_t payloads_ring_buffer[TRANSPORT_FIFO_MAX_FRAME_DATA];`
is moved to
`struct min_context`
so that each port has its own ring buffer.

The required changes are:
1. min.h - Add ring buffer in 
[image]

2. min.c - Remove old ring buffer
[image]

4. min.c - make sure the two usages of the old buffer access the new buffer instead
[image]
[image]

kinimodkoch, 2023-07-17:
I just realized this topic was also discussed in #46, but I still think this is a bug.
The comment in #46 is that the application is not designed to work with multiple ports - but apart from this minor issue, the multi-port support is implemented well.
I also don't agree with the concern about the usage of the protocol in tiny controllers - the suggested fix has no memory impact if you instantiate only one port.

kentindell, 2023-08-10:
The explanation is that I started to implement multiple ports but didn't complete. I'll accept a PR covering this, though.