min-protocol/min #52: What is the best practice for calling min_poll

pioneerAlone, 2024-01-09:
Hello ,

I am trying to port this library.
```
// Must be regularly called, with the received bytes since the last call.
// NB: if the transport protocol is being used then even if there are no bytes
// this call must still be made in order to drive the state machine for retransmits.
void min_poll(struct min_context *self, uint8_t const *buf, uint32_t buf_len);
```
```
#define TRANSPORT_FRAME_RETRANSMIT_TIMEOUT_MS       (50U) // Should be long enough for a whole window to be transmitted plus an ACK / NACK to get back
#define TRANSPORT_IDLE_TIMEOUT_MS                   (1000U)
```
**>> In my application scenario, min_poll() will be blocked until the serial port has data. But I think this will destroy the internal state machine. If it must be called periodically, what is the best delay time?**

kentindell, 2024-01-09:
The polling should be non-blocking (that's why it's designed to have 0 bytes returned).

The scheme was designed to fit into simple embedded main loop programs where the whole program is run periodically at (say) 10ms. Or in a task in an RTOS where some part of the application runs periodically. 

Typically you would have a UART interrupt handler to pick up bytes and put them into a buffer, then poll the buffer. The polling periodicity needed is derived from a combination of buffer size and baud rate, so that the polling is fast enough for the buffer not to overflow. So you can make the buffer big enough to fit a convenient polling period, or you can make a polling period fast enough to empty the buffer before overflow. I would shoot for something like a 10ms polling period and see how that fits.

pioneerAlone, 2024-01-10:
Thanks，
Another question，

If the read serial port fails, min_poll will poll the last data frame, but if the underlying serial port interface called by internal on_wire_bytes fails to send data, will this affect the internal state machine? Since the library does not have an error handling mechanism, I want to know this What should be done, such as calling min_transport_reset to reset the state machine

kentindell, 2024-01-10:
If the read fails then just have it produce no bytes. The protocol should catch that: the frame will fail its CRC because it will be missing data, the byte stuffing will cause a resync. The transport layer will spot the missing frame and retransmit.

closed 2024-01-11 by pioneerAlone